256dpi/arduino-mqtt #327: subscribe fails HARD

hurricaneJoef, 2024-06-23:
I noticed you have it set up if the mqtt client fails when subscribing to a topic (like a topic already subscribed to) it closes the connection to the server.

is this intentional?
if so can an alternative function be added to not close in this way?

(I have already fixed after a long effort server side this so I can submit a pr if it would be wanted)

wagfeliz, 2024-08-14:
I am getting this issue also...

256dpi, 2024-08-30:
Yes, until now we followed a fail hard mantra when it comes to protocol level errors. I'm happy to review pull requests that soften that a bit for more advanced use cases. May we can add a `client.setSoftFail(bool)` method that sets an internal variable that does not close the connection on such errors?