256dpi/arduino-mqtt #332: Subscribe is unreliable

Norbert445, 2024-09-06:
Hi, I'm using an ESP32 board, and when I subscribe to multiple topics, only a few subscription methods return a result, while the rest seem to get stuck in a deadlock. Sometimes only one subscription method works, and other times, more subscriptions succeed. If I disconnect from Wi-Fi, the stuck subscription methods return the result. With version 2.5.0, everything works smoothly, but any later version leads to these issues. How can I resolve this?

burner-, 2024-10-08:
Looks that also at arduino platform result is same. If I subscribe one it works fine. If I subscribe multiple topics arduino will reboot. But I was unable to get it work even with older versions. I also try add delay and process mesages between subscribes but no luck.

entrusc, 2024-12-21:
I second what @Norbert445 said: I'm also unable to more than one topic using 2.5.2 on an ESP32-S3. When I downgrade to 2.5.0 it works fine. Seems there were some bugs introduced in 2.5.1 or 2.5.2. Would appreciate if someone could take a look. Thanks.

entrusc, 2024-12-22:
Okay, after some more experimentation it turns out that 2.5.0 also has the same issue. It seems that subscribing to more than one topic sometimes works and sometimes doesn't, while subscribing to just one topic so far always works for me. Maybe the problem is caused by a networking problem though. I suggest to maybe implement some kind of re-try logic to make the subscription process more robust.