256dpi/arduino-mqtt #336: Sending floating point and integer data through MQTT

mitra42, 2024-09-28:
I've got some sensors generating floating point and integer data. I'd like to send this over MQTT, and ideally without converting back and forth to the Ascii. 

The docs have bool publish(const char topic[], const String &payload);
bool publish(const char topic[], const char payload[]);

I'm wondering if there is an accepted method of sending numbers (floating or int) over MQTT, and if so what it is.... I've tried searching for the answer in all the usual places, but I'm not finding it. 

I also think it would be useful to have this in at least one of the examples.

ForrestErickson, 2024-10-24:
Here is an answer the might work. Use JSON libraries to make a JSON object and then convert to binary to send.

The top answer here looks promissing.
https://stackoverflow.com/questions/23947779/how-to-send-data-as-json-objects-over-to-mqtt-broker

Let us know what does and does not work.

mitra42, 2024-10-25:
I ended up converting to a string - yes you can use JSON but its more overhead (code size) on small processors.

ForrestErickson, 2024-10-25:
Regarding, "I ended up converting to a string"
Thanks for letting us know you solution. 
**Can you post the code that works** so that persons who find this question, as I did, will find an answer?

For what it is worth I have no idea how memory safe the JASON solutions might be. I have read that Strings in Arduino are not safe under some programing methods because of limited memory.  Problem is more sever on an UNO of course.

ForrestErickson, 2024-10-25:
@mitra42 
Thanks for the example. I am some times novice and seeing the simple function
```
void messageSend(String &topic, float &value, int width, bool retain, int qos) {
  String *foo = new String(value, width);
  messageSend(topic, *foo, retain, qos);
}
```
helps

Just to be clear, since I see no conditional tests, you are sending every time a float and then an int.  Am I right?

And thanks again.

mitra42, 2024-10-25:
Sorry, no the width variable is misnamed, should be "decimalplaces" so `String(1.2345678,3)` becomes `"1.234"`

https://docs.arduino.cc/language-reference/en/variables/data-types/stringObject/