256dpi/arduino-mqtt #343: Large payload disconnects the client

Mis1eader-dev, 2025-04-06:
Sending a large payload, 2162 bytes as an example, disconnects the client immediately and reports `-6` as an error, which corresponds to:
```c++
LWMQTT_NETWORK_FAILED_WRITE = -6
```

Mis1eader-dev, 2025-04-06:
Upon a call to `mqtt.publish(...)`, I notice [this](https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.cpp#L94) line:
```c++
// write bytes
*sent = n->client->write(buffer, len);
```

First returning `25`, and then returns a `0` and ends up setting the error code to `LWMQTT_NETWORK_FAILED_WRITE`

joysfera, 2025-11-15:
For me, on ESP8266 it almost always disconnects while sending a large payload on the latest v2.5.2 with synchronous publish, while on ESP32 it's stable. Same source code. I've got delay(10) after client.loop(), no help.